raschy/ioBroker.solarmanpv #4: Installation geht nicht weiter

fr00sch, 2022-06-24:
Hey raschy,

vorab schonmal vielen Dank für den Adapter, leider war es zu Anfang etwas schwer den Adapter zu finden:
Der Adapter wird in dem Menü als "undefined" dargestellt:
[image]

Dann bleibt leider die Installation hier stehen und es gibt keine Infos über die Debug-Schnittstelle:
<html><body>

host.iobroker | 2022-06-24 15:18:14.212 | info | iobroker Installing raschy/ioBroker.solarmanpv... (System call)
-- | -- | -- | --
host.iobroker | 2022-06-24 15:18:14.211 | info | iobroker NPM version: 6.14.17
host.iobroker | 2022-06-24 15:18:14.004 | info | iobroker install raschy/ioBroker.solarmanpv
host.iobroker | 2022-06-24 15:18:13.390 | info | iobroker url raschy/ioBroker.solarmanpv --host iobroker

</body>
</html>

Meine Versionen sind:
Node.js v14.19.3 (Empfohlene Version v16.15.1)
NPM    6.14.17
iobroker-Version: 4.0.23

Gruß fr00sch

fr00sch, 2022-06-26:
Hey, ich habe sogar schon die Admin 6.1.4 und kann auch leider nicht mehr zurück.

Ilovegym66, 2022-06-26:
@fr00sch update mal auf die vorgeschlagene Node und NPM Version und versuche dann eine Installation mit einer funktionierenden Admin Version ( 6.1.2 etc, keine 6.1.4 ) nochmal.. :-)

raschy, 2022-06-26:
@fr00sch Ist denn der Adapter schon/trotzdem sichtbar? Ich hatte in einem älteren Testsystem auch so einen Effekt. Die Installation des Adapters blieb hängen, aber ich konnte auch bei nodejs 4.19.1, npm 6.14.16 und Admin v5.3.8 eine Instanz hinzufügen. Dann die richtigen Parameter und dann müsste es laufen.

fr00sch, 2022-07-08:
@raschy 
Mit einem Update bin ich schon ein paar Schritte weiter gekommen:
Node.js: v16.16.0
NPM: v8.11.0(8.12.1)
Admin-Adapter V6.1.11
[image]
Läuft die Install

closed 2022-07-08 by fr00sch

raschy, 2022-07-08:
@fr00sch Ja das freut mich. Die Warnungen kommen daher, dass das Repository noch nicht auf npm ist. Kommt aber auch noch. Viel Spaß.