raschy/ioBroker.solarmanpv #13: Question: Would it be possbile to get the data directly from the logger of the inverter?

der1sebi, 2022-08-29:
Do avoid the cloud from the solarman app and bringing my information to a China cloud it would be great to get the data directly from the inverter. Have you planned something like this for future updates?

raschy, 2022-08-29:
I am trying to read out the inverter directly in various ways. So far I have not really been successful.

der1sebi, 2022-08-30:
Thats great to hear. Normally it should be similar to the steps youre using now I guess?!

der1sebi, 2022-08-30:
Have found this topic on FHEM. Maybe it will help you?
https://forum.fhem.de/index.php/topic,123506.0.html

der1sebi, 2022-09-02:
And maybe this one will be very interesting for you as well:
https://github.com/dr-ni/mi600

raschy, 2022-09-02:
Many thanks for the tips. I have already studied the articles and a few others, but have not yet come to a satisfactory conclusion.

der1sebi, 2022-09-02:
Thanks for your effort. I really appreciate it.

raschy, 2022-09-05:
After extensive research, the implementation of the local readout is not quite trivial. On the other hand, this would be a completely different project than this one. Therefore, this issue is closed.

closed 2022-09-05 by raschy