raschy/ioBroker.solarmanpv #33: Doppelte Log-Einträge

unstressable, 2022-11-09:
Folgende Einträge erscheinen doppelt im Log:

`<html>
<body>

solarmanpv.0 | 2022-11-09 15:36:04.952 | error | [onReady] error: AxiosError: timeout of 2000ms exceeded
-- | -- | -- | --
solarmanpv.0 | 2022-11-09 15:36:04.948 | warn | [initializeStation] error: AxiosError: timeout of 2000ms exceeded
solarmanpv.0 | 2022-11-09 15:00:08.947 | error | [onReady] error: AxiosError: timeout of 2000ms exceeded
solarmanpv.0 | 2022-11-09 15:00:08.944 | warn | [initializeStation] error: AxiosError: timeout of 2000ms exceeded
solarmanpv.0 | 2022-11-09 14:54:06.228 | error | [onReady] error: AxiosError: timeout of 2000ms exceeded
solarmanpv.0 | 2022-11-09 14:54:06.222 | warn | [initializeStation] error: AxiosError: timeout of 2000ms exceeded

</body>
</html>`

raschy, 2022-11-09:
Welche Version läuft da?

unstressable, 2022-11-09:
> Welche Version läuft da?

0.1.2

raschy, 2022-11-09:
Gibt es im iobroker-Repository im latest. node.js 16.x sollte installiert sein.

unstressable, 2022-11-10:
> Gibt es im iobroker-Repository im latest. node.js 16.x sollte installiert sein.

muss ich  da meinen Adapter zuerst deinstallieren?
[image]

raschy, 2022-11-10:
Das wäre am sinnvollsten. Alles Alte raus und dann die Version 0.2.0 aus dem Latest installieren. Anschließend die Credentials eingeben, dann sollte der Adapter laufen.

unstressable, 2022-11-10:
> Das wäre am sinnvollsten. Alles Alte raus und dann die Version 0.2.0 aus dem Latest installieren. Anschließend die Credentials eingeben, dann sollte der Adapter laufen.

ok, werde ich probieren. Aktuell wird mir der Adapter nicht im ioBroker angeboten, kann aber vielleicht daran liegen, dass ich ihn damals von github installiert hatte.

raschy, 2022-11-11:
Ist auch das richtige Repository ausgewählt? > http://iobroker.live/repo/sources-dist-latest.json

unstressable, 2022-11-11:
Bei mir war die Stable drin:
http://download.iobroker.net/sources-dist.json

hab nun die latest hinzugefügt :
http://iobroker.live/repo/sources-dist-latest.json
und aktiviert, dann fand ich den Adapter.

Bin jetzt auf V0.2.0

Wieso ist er nicht im Stable drin?

raschy, 2022-11-11:
Nach welchen Kriterien der Adapter ins Stable kommt ist mir (noch) nicht bekannt. Wichtiger zu wissen wäre jetzt, ob der Adapter läuft!

unstressable, 2022-11-11:
Läuft nun, wenn wieder Fehler kommen, melde ich mich nochmal.

closed 2022-11-11 by unstressable