raschy/ioBroker.solarmanpv #34: Feature Request: Disable Auto-Refresh while sun is down

unstressable, 2022-11-18:
Would be nice to have a funktion to disable the Adapter after sunset and before sunrise to save machine power.

raschy, 2022-11-18:
Do you really think that turning off a scheduled adapter will save energy?

unstressable, 2022-11-18:
I think about computing power and a clean log (Axios error; timeout)

raschy, 2022-11-18:
I'll look at the development effort and any negative side effects. In the adapter, sunrise and sunset must still be calculated, which does not have much to do with the readout of the cloud. Therefore probably not.

unstressable, 2022-11-18:
Just provide us an Object to disable the data fetch, then we could do the rest over a Java/Blockly Script.
The adapter by itself does not have to be disabled.

raschy, 2022-11-18:
This object (obj.common.enabled = status; // enable/disable adapter) is built into every adapter by default. Every user is free to turn the adapter off and on again with a script as he or she sees fit.

unstressable, 2022-11-21:
I did it via Blockly:
`<xml xmlns="https://developers.google.com/blockly/xml">
  <variables>
    <variable id="Kin_d{P!1pKMo+5cZU`O">Instanz</variable>
    <variable id="b/PDv/D$qo]b|JPj@BAK">start</variable>
  </variables>
  <block type="procedures_defcustomnoreturn" id="`HD|n)F.zEhjAS[_Ha*9" x="13" y="13">
    <mutation statements="false">
      <arg name="Instanz" varid="Kin_d{P!1pKMo+5cZU`O"></arg>
      <arg name="start" varid="b/PDv/D$qo]b|JPj@BAK"></arg>
    </mutation>
    <field name="NAME">Adapter_start_stop</field>
    <field name="SCRIPT">ZXh0ZW5kT2JqZWN0KCdzeXN0ZW0uYWRhcHRlci4nICsgSW5zdGFueiwge2NvbW1vbjoge2VuYWJsZWQ6IHN0YXJ0fX0pOw==</field>
    <comment pinned="false" h="80" w="160">Beschreibe diese Funktion …</comment>
  </block>
  <block type="astro" id="xRiXJdFBL`%i79;vO~0H" x="12" y="88">
    <field name="TYPE">sunrise</field>
    <field name="OFFSET">-60</field>
    <statement name="STATEMENT">
      <block type="comment" id="J`iHK;E)]Y7Hpp|Mi7UZ">
        <field name="COMMENT">Instanz anschalten</field>
        <next>
          <block type="procedures_callcustomnoreturn" id="ih~h61Q(ObZv?~_lB4)0">
            <mutation name="Adapter_start_stop">
              <arg name="Instanz"></arg>
              <arg name="start"></arg>
            </mutation>
            <value name="ARG0">
              <block type="text" id="E2cJ`,2vK)}G-bva[Hws">
                <field name="TEXT">solarmanpv.0</field>
              </block>
            </value>
            <value name="ARG1">
              <block type="logic_boolean" id="f9;bz|~4KqI9RDyGrziD">
                <field name="BOOL">TRUE</field>
              </block>
            </value>
          </block>
        </next>
      </block>
    </statement>
    <next>
      <block type="astro" id="]jBWrp@QHd4y/?;S37gF">
        <field name="TYPE">sunset</field>
        <field name="OFFSET">60</field>
        <statement name="STATEMENT">
          <block type="comment" id="i[M?Qf5?2T~Y=QG*;i5{">
            <field name="COMMENT">Instanz ausschalten</field>
            <next>
              <block type="procedures_callcustomnoreturn" id="`WSmT8#_0n}j[9oERmhj">
                <mutation name="Adapter_start_stop">
                  <arg name="Instanz"></arg>
                  <arg name="start"></arg>
                </mutation>
                <value name="ARG0">
                  <block type="text" id="M:`#C-:](kJ2YQC2zc`q">
                    <field name="TEXT">solarmanpv.0</field>
                  </block>
                </value>
                <value name="ARG1">
                  <block type="logic_boolean" id="TmEnz*n8;kq3M:d6%H%h">
                    <field name="BOOL">FALSE</field>
                  </block>
                </value>
              </block>
            </next>
          </block>
        </statement>
      </block>
    </next>
  </block>
</xml>`

[image]

closed 2022-11-21 by unstressable