raschy/ioBroker.solarmanpv #43: Fehler im Log, wenn PV Anlage keine Erzeugung liefert

unstressable, 2022-12-02:
Mir ist aufgefallen, dass der Adapter immer folgenden Fehler ausgibt, wenn die PV Anlage keinen Strom erzeugt.
Passiert, wenn Sonne untergegangen ist, das konnte ich in den Griff bekommen, indem ich den Adapter bei Sonnenuntergang deaktiviere.

Leider kommt es aber auch bei starker Bewölkung mitten am Tag vor:

`<html>
<body>

solarmanpv.0 | 2022-12-02 14:10:16.748 | error | DB closed
-- | -- | -- | --
solarmanpv.0 | 2022-12-02 14:10:16.747 | error | Error: DB closed at close (/opt/iobroker/node_modules/ioredis/built/redis/event_handler.js:184:25) at Socket.<anonymous> (/opt/iobroker/node_modules/ioredis/built/redis/event_handler.js:151:20) at Object.onceWrapper (node:events:628:26) at Socket.emit (node:events:513:28) at TCP.<anonymous> (node:net:301:12)
solarmanpv.0 | 2022-12-02 14:10:16.746 | error | unhandled promise rejection: DB closed
solarmanpv.0 | 2022-12-02 14:10:16.745 | error | Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().

</body>
</html>`

raschy, 2022-12-02:
Ich habe mehrere Instanzen von SolarmanPV laufen. Die Anlagen sind in ganz Deutschland verteilt und haben derzeit alle nicht viel Sonne. Bei keiner der monitorten Anlagen tritt dieser Fehler auf. Da ich diesen Fehler nicht reproduzieren kann sehe ich mich derzeit außerstande, diesen Fehler abzustellen.

unstressable, 2022-12-02:
Hmmm, schade. Dann warten wir mal ab.

raschy, 2022-12-02:
Ich werde weiter beobachten, ob und wann dieser Fehler kommt. Möglicherweise hat er direkt mit dem Adapter nichts zu tun

raschy, 2023-02-13:
Sollte in der aktuellen Version 0.2.2 nicht mehr vorkommen.

closed 2023-02-13 by raschy

unstressable, 2023-02-13:
okay, das werde ich beobachten. Danke für deine Entwicklerarbeit!

unstressable, 2023-02-13:
@raschy kommt wohl noch immer vor. Heute nach Sonnenuntergang:
`<html>
<body>

solarmanpv.0 | 2023-02-13 18:30:13.090 | warn | [initializeStation] error: AxiosError: timeout of 4000ms exceeded
-- | -- | -- | --

</body>
</html>`

raschy, 2023-02-13:
Ich nehme das zur Kenntnis und forsche weiter.

unstressable, 2023-02-13:
Danke!