raschy/ioBroker.solarmanpv #50: Warnung bei Adapterstart mit Wechselrichter Deye Sun-800G3

Stefan-3, 2023-01-31:
Adapter SolarmanPV version 0.2.0 vom 7.11.2022

Bei Verwendung des 2x400W Wechselrichters "Deye Sun-800G3-EU-230" erscheinen im IoBroker Log die Warnings:

-------------------------------------------------------------------------------------------------------------------------------
Object solarmanpv.0.3104772.221753192 is invalid: obj.common.name has an invalid type! Expected "string" or "object", received "number"
-------------------------------------------------------------------------------------------------------------------------------
This object will not be created in future versions. Please report this to the developer.
-------------------------------------------------------------------------------------------------------------------------------

Das Auslesen des Wechselrichters funktioniert jedoch einwandfrei. Vielen herzlichen Dank für den tollen Adapter 🙂

raschy, 2023-01-31:
Kommt die Meldung regelmäßig oder kam die nur beim ersten Start?

Stefan-3, 2023-01-31:
Die Meldung kam nur einmalig beim allerersten Start. Selbst wenn man den Adapter stoppt und wieder startet kommt die Meldung nicht mehr. Von meiner Seite aus können wir diesen issue auch wieder schließen. Ich war nur so happy über deinen tollen Adapter, so dass ich mich verpflichtet gefühlt habe "der Anweisung im Log Folge zu leisten" ;-)
Die beiden Meldungen aus diesem issue kamen um 16:42 Uhr und um 17:00 Uhr kam die timeout Meldung ([initializeStation] error: AxiosError: timeout of 2000ms exceeded). Vielleicht hat der Wechselrichter um 16:42 schon ausgeschalten. Ich schlage vor ich untersuche das die nächsten Tage nochmal und gebe dann Bescheid.

raschy, 2023-02-01:
OK, mach das so.

Stefan-3, 2023-02-06:
So, jetzt bin ich endlich dazu gekommen. Geht ja nur wenn die Sonne scheint und da hat man im Winter etwas weniger davon ;-)
Vorgehensweise:
Kontrolliert ob PV Anlage an ist >>> Instanz des Adapter gelöscht >>> IoBroker mittels PUTTY gestoppt dann wieder gestartet >>> Adapter installiert, konfiguriert und dann gestartet
Ergebnis:
Beim Erstmaligen Start des Adapters erscheinen die beiden Meldungen (... 0.3104772.221753192 is invalid ... bzw. ...report this to the developer).
Ich habe den Test 2x durchgeführt mit identischem Ergebnis.
Vollständiger Log:
[Log Reproduzierung.pdf](https://github.com/raschy/ioBroker.solarmanpv/files/10661558/Log.Reproduzierung.pdf)

raschy, 2023-02-06:
Danke für die Rückmeldung. Es klingt danach, dass beim ersten Aufbau des Objektbaums einmal eine Anweisung nicht sauber ist. Hier wird ein anderer Wert erwartet als der Adapter schreibt. Ich werde versuchen, das zu fixen.

Stefan-3, 2023-02-12:
Ich hab die neue Version 0.2.2 des Adapters heute ausprobiert (auf gleiche Weise wie zuletzt auch) und nun erscheint keine Fehlermeldung mehr im Log :-) Respekt, dass Du das Thema so schnell bearbeitet und gelöst hast. Vielen herzlichen Dank.

closed 2023-02-12 by Stefan-3

raschy, 2023-02-12:
Danke für die positive Rückmeldung. Dann können wir diesen Issue jetzt schließen.

Stefan-3, 2023-02-12:
Jawolla, hab ich bereits gemacht. Oder ist das unüblich? Schließt üblicher Weise der Implementierer solche Issues? Oder gibt es da keine feste Regel? Das war mein allererster Issue den ich gestellt habe, daher mein Nachfragen zur Vorgehensweise.

raschy, 2023-02-12:
Yes Sir, alles ok. Sollte dann nur nicht offen bleiben.