raschy/ioBroker.solarmanpv #51: Timeout Warning alle 30 Minuten

Stefan-3, 2023-01-31:
Adapter SolarmanPV V0.2.0 vom 7.11.2022
Wechselrichter Deye Sun-800G3-EU-230

Sobald die Sonne weg ist schaltet dieser Wechselrichter seine WLAN-Verbindung aus. Ich vermute, dass dies die Ursache ist, warum dann zyklisch folgende Warning im IoBroker Log steht:

[initializeStation] error: AxiosError: timeout of 2000ms exceeded
(Meldung um 17:00, 18:00, 21:00, 21:30 Uhr)

Zusätzlich kommt vereinzelt noch folgende Warning:

[getDeviceData] error: AxiosError: timeout of 2000ms exceeded
(Meldung um 18:06 Uhr)

raschy, 2023-01-31:
Wenn nicht genug Sonne scheint wird die interne Elektronik nicht mit Spannung versorgt - ergo: kein WLAN. Die Meldung von Axios (in Maßen) ist auch normal, wenn die Antwort vom Server in China zu lange dauert. Ich hatte heute auch ca. 8 dieser Meldungen im Log. Ich kann die Warnung zwar rauswerfen, aber damit ist der Sache auch nicht geholfen.

Stefan-3, 2023-01-31:
Aha, und ich dachte das würde nur mit meinem 800W Wechselrichter auftreten, da ich über Google nicht wirklich etwas zu der IoBroker Meldung gefunden habe. Vielleicht sollte man die Meldung als Info deklarieren anstatt als Warning. Warnings sehen im IoBroker Log immer so "schlimm" aus und wenn es jede Nacht mehrfach vorkommt, besteht die Gefahr, dass man "echte" IoBroker Warnings übersieht. Ggf. kann ich aber auch damit leben so wie es jetzt ist.

raschy, 2023-02-01:
Warnungen sehen tatsächlich etwas bedrohlich aus. Ist halt eine Warnung, dass die Werte in den Datenpunkten eventuell (oder bestimmt) nicht mehr den wirklichen Wert beinhalten.

Stefan-3, 2023-02-13:
Mit der neuen Version des Adapters (SolarmanPV V0.2.2) gibt es die Timeout-Warnings nicht mehr. Finde ich gut so. Man kann ja die Datenpunkte des Adapters immer auslesen (egal ob der Wechselrichter an oder aus ist). Und wenn man sicherstellen will, dass die Datenpunkte aktuell sind, dann muss man eben den Datenpunkt connectStatus=1 überprüfen.
Was durchaus Sinn machen würde, dass man die Timeout Warning ausgibt, wenn connectStatus=1. Dann ist nämlich wirklich etwas faul. Vielleicht ist das so aber jetzt auch schon umgesetzt.
Das ursächliche "Problem" ist jedenfalls gelöst, so dass man den Vorgang schließen kann. Vielen Dank nochmal für die tolle und schnelle Unterstützung :-)

closed 2023-02-13 by Stefan-3