raschy/ioBroker.solarmanpv #55: Feature Request: extract total production

snoerfsnoerf, 2023-02-19:
Thanks for the create plugin - works perfect. 

Is it possible to extend the feature by extracting additional information provided by the API? especially the "Total Production" would be a valuable field for me. 
Thanks Dirk

Baenker, 2023-02-19:
Verstehe leider nicht genau was Du meinst?  Es gibt doch z. B. solarmanpv.0.xxxxxx.Et_ge0 das ist doch Total Production?

raschy, 2023-02-19:
In the meantime, my horizon has also expanded somewhat and I can now not only collect relevant data from the inverters, but also virtually all values from batteries and collectors. This will be reflected in the upcoming version 0.3.1.

raschy, 2023-03-28:
No new findings or requirements

closed 2023-03-28 by raschy