raschy/ioBroker.solarmanpv #59: Keine Verbindung trotz Dateneingabe

Peterlustig123456, 2023-03-12:
Hallo zusammen, 

die Daten habe ich ohne Probleme von SOLARMAN bekommen. 

1.Zeile ID 2.Zeile Geheimniss

leider keine Daten bekommen, gibt es hier weiter etwas zu beachten? 

Danke im Voraus!

raschy, 2023-03-12:
Hallo Peterlustig123456, wie kommst du auf 1.Zeile / 2.Zeile? Bei mir sieht das UI so aus und sollte mit den richtigen Daten von dir auch funktionieren.
[image]

Peterlustig123456, 2023-03-12:
Hallo raschy,

ja richtig ist nebeneinander, jedoch bekomme ich in den Objekten keinerlei Daten angezeigt.
Bin da noch recht neu in der Materie, bitte um Nachsicht.

APP ID sowie APP Secret PW und Mail ist eingetragen

Gruß

raschy, 2023-03-12:
Evtl. blöde Frage: scheint die Sonne? Was steht im Protokoll unter 'solarmanpv.0' ?

Peterlustig123456, 2023-03-12:
sonne ist ausreichend,WR läuft.

anbei mal das Protokoll
[image: Bildschirmfoto 2023-03-12 um 14.18.40.png]

raschy, 2023-03-12:
Leider kein Foto

Peterlustig123456, 2023-03-12:
hoffe hat jetzt geklappt

raschy, 2023-03-12:
Ich vermute mal, dass das mit dem Hochladen des Fotos nicht so ganz klappt. Das Foto in den unteren Bereich "Attach files by dragging ... " reinziehen.

Peterlustig123456, 2023-03-12:
[image]

raschy, 2023-03-12:
Ok. Jetzt kann ich was sehen. Was zeigt denn die Solarman App an?  Offensichtlich sind deine Credentials ok, aber der China-Server antwortet nicht (richtig).

Peterlustig123456, 2023-03-12:
Die App funktioniert regulär, kann ich in der App zusätzlich etwas prüfen?

raschy, 2023-03-12:
Dann wird es für mich etwas schwierig. Traust du mir, dann schick mir deine Zugangsdaten per Mail an raschy@gmx.de. Dann kann ich etwas tiefer nachschauen.

FabulousGee, 2023-03-23:
Habe leider das selbe Problem

[image]

raschy, 2023-03-23:
Das Problem ist wahrscheinlich ähnlich. Zum einen sieht es so aus, als ob der Adapter mehrfach laufen würde "solarmanpv.0 already running". 
Für den Fehler "[initializeStation] error: undefined" würde ich die Cloud verantwortlich machen. Meiner Beobachtung nach ist die Solarman-Plattform derzeit sehr instabil. Daher schlage ich vor, die nächsten Tage abzuwarten.

Pixelshading, 2023-03-23:
[image]

Jup hab auch das selbe problem

raschy, 2023-03-24:
@Peterlustig123456 @Pixelshading Anhand der Daten von FaboulosGee habe ich feststellen müssen, dass mit den angegebenen Credentials von der Solarman-Cloud kein Token abgeholt werden konnte. Daher habe ich den Vorschlag gemacht, beim Solarman-Support anzufragen, ob da etwas schief läuft. Eine Anfrage dazu könnte so lauten: "I am a user of the Solarman app and have also been retrieving data via the Api using Api ID and Api Secrets for over half a year. Since yesterday, I no longer receive a valid token or an error message "could not retrieve token". Do you have an explanation for this or how can we remedy the situation?"

Peterlustig123456, 2023-03-26:
@raschy vielen Dank es funktioniert endlich !

closed 2023-03-26 by Peterlustig123456

raschy, 2023-03-26:
@Peterlustig123456 Darf ich noch für die anderen User fragen: Hast du den Support angemailt und was ist danach passiert? Neue Credentials oder nur ein "Erledigt" aus China?

Peterlustig123456, 2023-03-26:
Weder noch, hatte die Zugangsdaten nochmals eingegeben, dann einen Tag gewartet. Jetzt passt es.

FabulousGee, 2023-03-29:
Bei mir hat sich soeben herausgestellt, dass die API-ID wohl falsch übermittelt wurde. Der Herr vom Support hat hinten eine Zahl zu wenig übermittelt :-) Bei mir geht es nun auch...

War wohl eine Kombination aus API-Problemen und (bei mir) Credentials.