raschy/ioBroker.solarmanpv #68: Blacklisting does not work

unstressable, 2023-04-20:
Could anyone please explain the blacklisting feature?

I got 2 different PV projects over the same account.
[image]

I want to see them respectively alone on two different ioBroker installations.

When I use the phrase "1918486" for example nothing changes.

raschy, 2023-04-20:
The blacklist refers to individual data points below an object tree. In this case, both folders are supplied by one account and cannot be split.
Hiding a folder if it is not of interest is planned for a later version.

closed 2023-04-26 by raschy

simplest-circuits, 2023-05-20:
Can you share how to use the "blacklist" in general, please?

raschy, 2023-05-21:
The 'Blacklist' is designed to hide individual data points that are not needed. 
To do this, enter the terms here separated by commas and then delete the data point manually.
[image]