raschy/ioBroker.solarmanpv #79: Moduls not updated

sokrates-sith, 2023-06-01:
Since the last update the modules are not updated anymore.
I have v0.4.2

[image]

only the whole power of the system is shown correct

raschy, 2023-06-01:
Are the modules for evaluation also ticked on the 'System modules' tab?
[image]

sokrates-sith, 2023-06-02:
ah, nope, the tab is empty. anything i can do to load them?
[image]

Ilovegym66, 2023-06-02:
hier auch so, es werden keine Untermodule angezeigt..

raschy, 2023-06-02:
Besteht die Möglichkeit, den Adapter mal komplett zu deinstallieren und dann wieder neu zu installieren?

sokrates-sith, 2023-06-02:
ok, das hat geholfen. ich sehe nun die module in "systemmodule" und aktuell werden die werte wieder aktualisiert.

raschy, 2023-06-02:
Danke für die Rückmeldung. Freut mich, dass es wieder funktioniert.

sokrates-sith, 2023-06-02:
danke auch für die schnelle hilfe

closed 2023-06-02 by sokrates-sith