raschy/ioBroker.solarmanpv #97: DP wechselt permanent von "true" auf "(null)"

Kusi-Schweiz, 2023-09-05:
Der Datenpunkt "system.adapter.solarmanpv.0.connected" wechselt vor jedem "CRON-Start" von "true" auf "leer", dann kurz "(null)", danach wieder auf "true".
Kann es geändert werden, dass dieser DP permanent auf "true" bleibt?

[image]
[image]
[image]

Die History vom DP "system.adapter.solarmanpv.0.connected":
[image]

Hintergrund ist der, dass ich  auf "IDs vom Selektor (system.adapter.*.0.connected)" triggere, um eine Meldung zu erhalten, wenn ein Dienst/Gerät nicht mehr erreichbar ist, respektive die Verbindung wieder hergestellt wurde.

Das Blockly:
[image]

raschy, 2023-09-06:
Der Adapter ist ein 'scheduled Adapter', der während der inaktiven Zeit nicht läuft. Dieses 'connected' prüft also nur die Verbindung vom Admin (js-controller) zum Adapter und wird also ständig zwischen (null) und true wechseln. Um das oben beschriebene Vorhaben über diesen Weg zu lösen, müsste ich den Adapter als Daemon umschreiben - das ist aber nicht geplant.

Kusi-Schweiz, 2023-09-07:
Danke für die Information. Dann werde ich diesen Abdapter bei der "connected"-Abfrage vorübergehend herausnehmen mit einem "falls"-Block.

Könnte denn programmiert werden, dass dieser Datenpunkt nur "true" und "false" hat?

closed 2023-10-01 by raschy