raschy/ioBroker.solarmanpv #109: Log Spam "has to be number, but received string"

unstressable, 2023-10-28:
Hi,

bei jedem geplanten Adapterstart wird das Log mit folgenden Infos gefüllt:
`
solarmanpv.0
2023-10-28 12:00:15.803	info	State value to set for "solarmanpv.0.xxx.xxx.DEV_TYP1" has to be type "number" but received type "string"

solarmanpv.0
2023-10-28 12:00:14.889	info	State value to set for "solarmanpv.0.xxx.xxx.ST_w1" has to be type "number" but received type "string"

solarmanpv.0
2023-10-28 12:00:14.820	info	State value to set for "solarmanpv.0.xxx.xxx.NOP" has to be type "number" but received type "string"

solarmanpv.0
2023-10-28 12:00:14.450	info	State value to set for "solarmanpv.0.xxx.xxx.HWv1" has to be type "string" but received type "number"`

raschy, 2023-10-28:
Hier hat Solarman die Ausgabewerte Ihrer API geändert. Einfach in den Datenobjekten die angemeckerten Datenpunkte löschen. Die werden dann von dem Adapter wieder richtig angelegt.

unstressable, 2023-10-28:
Adapter stoppen, alles löschen und wieder starten? Geht das so auch?

raschy, 2023-10-28:
Ja, geht natürlich auch - ist aber unnötig. Daran denken, wenn Datenpunkte aufgezeichnet werden, die wieder zu aktivieren.

unstressable, 2023-10-28:
hat geklappt, waren etliche Datenpunkte, die moniert wurden, hab alles gelöscht und neu anlegen lassen.
Danke!

closed 2023-10-28 by unstressable