raschy/ioBroker.solarmanpv #157: B_Pdcg1 (Akku lade/entlade Leistung) wird nicht mehr aktualisiert

ltspicer, 2024-05-13:
B_Pdcg1 (Akku lade/entlade Leistung) wird seit heute Morgen 4:00 Uhr nicht mehr aktualisiert.
Die anderen Datenpunkte funktionieren normal.

Habt ihr das Problem auch?
Ist das nur ein vorübergehendes Problem oder sendet der Solarman Server diesen Wert unter einem anderen Namen?

**Versions:**  
 - Adapter version: v0.5.1
 - JS-Controller version: 5.0.19
 - Node version: v18.20.2
 - Operating system: Debian Bullseye (on Raspberry)

raschy, 2024-05-13:
Wenn Solarman Werte ändert, sollten sie bei Eintreffen eines gültigen Wertes diesem mit dem 'neuen' Namen im Objektbaum anlegen. (Ist nicht von der Hand zu weisen, weil es in der Vergangenheit auch schon mal Typänderungen gab. So wurde aus einem numerischen Wert ein String. Das muss allerdings manuel angepasst werden.) Ich selbst habe keinen Akku und kann daher zum Grundproblem nichts sagen.

ltspicer, 2024-05-13:
Würde sich das Änderungsdatum nicht ändern, wenn plötzlich ein String anstelle eines numerischen Wertes daher kommt?
[image]

ltspicer, 2024-05-13:
Wirklich merkwürdig. Die meisten anderen Daten kommen an.
Auf der Handy-App kommen die aktuellen Akku-Daten auch an.
Hier nochmal ein Screenshot.

[image]

raschy, 2024-05-13:
Wenn neue Daten kommen, ändert sich natürlich auch das Datum. Da alle anderen Werte kommen würde ich mal bis morgen warten, je nachdem, was mit dem Akku so passiert.

ltspicer, 2024-05-14:
Leider unverändert.
Ist hier niemand, der auch ein Akku hat?
Wenn ja, bitte mal nachschauen, ob der B_Pdcg1 sich aktualisiert.

ltspicer, 2024-05-14:
Kann es sein, dass der B_Pdcg1 durch B_P1 abgelöst wurde?

[image]

raschy, 2024-05-14:
Nichts ist unmöglich! Wie gesagt, ich habe keinen Akku und kann daher nichts dazu sagen. Bitte beobachten und berichten.

ltspicer, 2024-05-15:
Bis jetzt sieht es so aus, dass neu wirklich **B_P1** der gewünschte Datenpunkt ist. **B_Pdcg1** ist verwaist.
Mit dieser Erkenntnis schliesse ich dieses Issue mal ;)

closed 2024-05-15 by ltspicer