raschy/ioBroker.solarmanpv #171: No data since 11.07.2024 9:35h

unstressable, 2024-07-15:
Nothing changed, log says:
<html>
<body>

solarmanpv.0 | 2024-07-15 20:21:20.690 | warn | [initializeStation] error: undefined
-- | -- | -- | --

</body>
</html>

Adapter 0.5.1

ioBroker latest

raschy, 2024-07-15:
It looks like Solarman has changed their regulations and are actively seeking approval for the new developer regulations.
Please write to Solarman support. They will then reply with an appropriate reply form.

unstressable, 2024-07-22:
I wrote them, but no helpful answer.
They told me, that they changed the API System, but they did not tell how to fix the problem.

raschy, 2024-07-22:
The dialog with Solarman Support seems to be a little bit slow.  
See also: https://forum.iobroker.net/topic/55799/solarman-pv-bosswerk-mi-deye/554

unstressable, 2024-07-24:
I switched to the "deyedic" Adapter. This one seem to work actally.
it seems to collect data locally. This is much better than via cloud.

raschy, 2024-07-24:
Thank you for your feedback. Have fun.

closed 2024-07-24 by raschy