raschy/ioBroker.solarmanpv #220: Update stable version in repo from 0.6.2 to 0.6.3

GermanBluefox, 2025-01-18:
# Think about update stable version to 0.6.3
**Version**: stable=**0.6.2** (70 days old) => latest=**0.6.3** (15 days old)
**Installs**: stable=**620** (51.03%), latest=**74** (6.09%), total=**1215**

Click to use [developer portal](https://www.iobroker.dev/adapter/solarmanpv/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L2502)

**Do not close this issue manually as a new issue will be created if condition for update still exists.**

Please drop a comment if any reason exists which blocks updating to version 0.6.3 at this time.

Note: This is an automatically generated message and not personally authored by bluefox!
      @mcm1957 for evidence

raschy, 2025-02-06:
Setting version 0.6.3 as stable did not work, so a new version was created.

closed 2025-02-06 by raschy

mcm1957, 2025-02-06:
????

Atleast manual adding to sources-dist-stable.json should always work.