raschy/ioBroker.solarmanpv #239: Repeating LOG Entries on Adapter Query

megamacy, 2025-03-08:
Hi, thanks for the adapter, really helpful!

Is there a way to disable the repeated log entries when the adapter starts fetching new values? My log is almost completely filled with these entries.

`host.iobroker | 2025-03-08 06:48:33.025 | info | instance system.adapter.solarmanpv.0 in version "0.6.4" started with pid 6017
host.iobroker | 2025-03-08 06:42:41.389 | info | instance system.adapter.solarmanpv.0 having pid 5716 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
solarmanpv.0 | 2025-03-08 06:42:40.832 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Everything done. Going to terminate till next schedule
solarmanpv.0 | 2025-03-08 06:42:34.423 | info | starting. Version 0.6.4 in /opt/iobroker/node_modules/iobroker.solarmanpv, node: v20.18.3, js-controller: 7.0.6`

raschy, 2025-03-08:
If these log entries interfere, you can set the log level of the adapter from “info” to “warning”.

megamacy, 2025-03-08:
This may be a workaround for the adapter itself, but set the host to another loglevel woul dinterfere with visibility of wanted entries. Maybe theres a way to run the adapter without cron?

raschy, 2025-03-08:
Sorry, I don't understand the question. The adapter must retrieve the data from the cloud independently. How is this supposed to work without time control (=cron)?

megamacy, 2025-03-08:
Other Adapters also running frequently without having a planned cron at the iobroker host. I think they running permenently and use internal timers?

raschy, 2025-03-08:
It may be that other adapters handle this differently. This adapter is designed as a “scheduled” adapter and relies on the time interval being set.

closed 2025-04-23 by raschy