raschy/ioBroker.solarmanpv #252: 0.6.4 works. 0.7.0+ not

ltspicer, 2025-05-27:
**To Reproduce**  
Steps to reproduce the behavior:
1. Install 0.7.1 (or 0.7.0)
2. See the errors in log (initializeStation) & lastUpdateTime will not be updated

**Screenshots & Logfiles**  
[image]
[image]

**Versions:**  
 - Adapter version: 0.7.0+
 - JS-Controller version: 7.0.6
 - Node version: v20.18.2
 - Operating system: Raspberry OS (RPI5)

raschy, 2025-05-28:
Sorry, but I can't do anything with this little information. I am using version 0.7.1 and it provides the data. The cloud may not be responding.

[image]

ltspicer, 2025-05-28:
Hmm, I install again the newest version.
I get

[image]

Now I downgrade to 0.6.4

raschy, 2025-05-28:
Ok, can you please send me your access data. Not here, of course, but (if you like) by e-mail raschy@gmx.de

ltspicer, 2025-05-28:
Done. See email ;)

ltspicer, 2025-05-28:
Do you need the log data?

raschy, 2025-05-28:
Fehler gefunden: Seit der Version 0.7.x werden email und App-Secret verschlüsselt gespeichert (ReadMe=> email and appsecret is now encrypted, please re-enter!) Somit werden beim Start wirre Zeichen eingelesen.
Lösung: Bitte Passwort und App-Geheimnisse löschen und neu eingeben.